cpldcpu/light_ws2812 #19: attiny13a arduino ide, 9.6mhz 1200000

FreezyExp, 2015-02-07:
So I gave this a shot:

    attiny13at9.name=ATtiny13 @ 9.6MHz (internal 9.6 MHz clock)
    attiny13at9.bootloader.low_fuses=0x7a
    attiny13at9.bootloader.high_fuses=0xff
    attiny13at9.upload.using=arduino:arduinoisp
    attiny13at9.upload.maximum_size=1024
    attiny13at9.build.mcu=attiny13
    attiny13at9.build.f_cpu=1200000
    attiny13at9.build.core=core13

But I guess the timings have not been updated for arduin yet.
Without knowing the exact workings of the code in the 'ws2812_sendarray_mask'
I would guess this needs attiny bit of love (see what I did there? ;-) )

FreezyExp, 2015-02-07:
I think I found the porblem, missing a 0 behind f_cpu

closed 2015-02-07 by FreezyExp

cpldcpu, 2015-02-08:
Ok, looks like you figured it on your own.

In general: The library adapts to the clock automatically. You just have to make sure F_CPU is set correctly and that the device is clocked at >4 Mhz, or better, >8 MHz.

FreezyExp, 2015-02-08:
Ended up adapting the bare AVR code as otherwise it would not fit on the tiny.
Did not take to long, but currently sitting at 828 bytes