cpldcpu/light_ws2812 #23: not working with ATMega8

escoand, 2015-04-23:
I tried the RGB_blinky example with my ATMega8 and the first LED is showing green - constantly.
I only had to change to small parts...

```
diff --git a/light_ws2812_AVR/Examples/RGB_blinky.c b/light_ws2812_AVR/Examples/RGB_blinky.c
index 1429d59..44ab9d5 100644
--- a/light_ws2812_AVR/Examples/RGB_blinky.c
+++ b/light_ws2812_AVR/Examples/RGB_blinky.c
@@ -21,8 +21,8 @@ int main(void)
   CCP=0xD8;            // configuration change protection, write signature
   CLKPSR=0;            // set cpu clock prescaler =1 (8Mhz) (attiny 4/5/9/10)
   #else
-  CLKPR=_BV(CLKPCE);
-  CLKPR=0;                     // set clock prescaler to 1 (attiny 25/45/85/24/44/84/13/13A)
+  //CLKPR=_BV(CLKPCE);
+  //CLKPR=0;                   // set clock prescaler to 1 (attiny 25/45/85/24/44/84/13/13A)
   #endif
 
 
diff --git a/light_ws2812_AVR/Makefile b/light_ws2812_AVR/Makefile
index ff0f6e6..e04009d 100644
--- a/light_ws2812_AVR/Makefile
+++ b/light_ws2812_AVR/Makefile
@@ -4,7 +4,7 @@
 # Change these parameters for your device
 
 F_CPU = 16000000
-DEVICE = attiny85
+DEVICE = atmega8
 
 # Tools:
 CC = avr-gcc
```

cpldcpu, 2015-04-24:
Well, it is actually noted in the source, that these parts have to be changed for a different CPU. So I am not sure what the issue is?

escoand, 2015-04-24:
Ok, not an issue more a helpless cry for some information. :-) Can you give me a hint how I can fix the constantly glowing LED? The CLKPR setting does not exist for ATmega8.

cpldcpu, 2015-04-24:
The first thing to check is if your cpu is clocked correctly. are you using an external crystal? check fuse and multiplier settings.

closed 2015-05-01 by cpldcpu