cpldcpu/light_ws2812 #33: Working on STM32F030F4P6 (ARM)

DedeHai, 2016-08-21:
Hi @cpldcpu 

just tested your code on an STM32F0, works like a charm.
Thank you for the code, I always wanted to write it myself but I am rubbish at assembler programming. 

Here is what I added to the header file:

`#include "stm32f0xx.h"			// Change this to the include for your MCU
#include "stm32f0xx_gpio.h"`

...

`#define ws2812_port_set ((uint32_t*)&GPIOA->BSRR)	// Address of the data port register to set the pin
#define ws2812_port_clr	((uint32_t*)&GPIOA->BRR)	// Address of the data port register to clear the pin

#define ws2812_mask_set  GPIO_Pin_3		// Bitmask to set the data out pin
#define ws2812_mask_clr  GPIO_Pin_3		// Bitmask to clear the data out pin`

cpldcpu, 2016-11-07:
Excellent, thanks for letting me know.

Oliv4945, 2016-12-08:
Maybe this issue can be closed regarding #38  ?

cpldcpu, 2016-12-09:
yes, i agree!

closed 2016-12-09 by cpldcpu