cpldcpu/light_ws2812 #34: Would this work with WS2813?

briceparent, 2016-10-07:
The new product WS2813 seems to be based on WS2811 and WS2812, both supported by this software. 
Is there (or will there be) any support for this 4-wire hardware?

I've ordered some strips, but I won't get them until some weeks, so I can't really make any test by now, but I'm already looking for solution (like an Attiny85 + this lib) to interface those strips with a Raspberry Pi.

cpldcpu, 2016-10-08:
My WS2813 order also has not arrived yet.

From the datasheet it looks as if the protocol remained the same, so it should work with the one wire protocol as well. 

Extension to two wires would be required to take into account the situation where the first LED in the chain fails.

Soldia1138, 2016-11-08:
Shouldn't be a problem if I interpret the dstasheet correctly. The second line is only for Backup Data.

cpldcpu, 2016-12-11:
I tested it now. It works. The only modification is an increased reset time.

closed 2016-12-11 by cpldcpu