cpldcpu/light_ws2812 #35: How to configure SET & CLR for SAMD21E15B?

RojasMilo, 2016-10-12:
Hi I'm trying to use in an SAMD21E15B 

[image]

But I do not know how to write the address of the data port register to set the pin, I'm using the PA14 pin

[image]

Thank you.

Soldia1138, 2016-11-08:
There are two errors in your line:
`PORT->Group[0]->OUTSET.bit[14]` isn't the correct syntax.

First of all the _OUTSET_ is part of the _Group_ *struct*, so it should be
`PORT->Group[0].OUTSET`

The next one is the bit address. You need to either address the bit (_bit_) itself or the whole register (_reg_). Technically it doesn't really matter, because it is a union the definition is:
```
typedef union {
  struct {
    uint32_t OUTSET:32;        /*!< bit:  0..31  Port Data Output Value Set         */
  } bit;                       /*!< Structure used for bit  access                  */
  uint32_t reg;                /*!< Type      used for register access              */
} PORT_OUTSET_Type;
```

So you can either type:
`PORT->Group[0].OUTSET.reg = PORT_PA14;`
or
`PORT->Group[0].OUTSET.bit.OUTSET = PORT_PA14;`

cpldcpu, 2017-01-14:
Thanks to soldia for solving this. Closing.

closed 2017-01-14 by cpldcpu