cpldcpu/light_ws2812 #37: Atmega328p bad timing

neoxnitro, 2016-11-12:
Hi, i tested with an atmega328p @20 MHz (external crystal) and the output timing are bad:

I confirmed that my AVR work at 20 MHz because the _delay_us work well and the serial baud rate work well too.

fuse: L: 0xFF | H: 0xD1 | E: 0x07 ( CKDIV8 is disabled)
#define F_CPU 20000000UL

USB analyser : 
0 CODE:
time high:  1.5 µs
time low: 3.25 µs
1 CODE:
time high:  3.25 µs
time low: 1.98 µs

i modified the following line to work perfectly (all example work well) but the timing results are not perfect (the values "1000000" was modified by "100000")

light_ws2812.c
```
#define w_zerocycles    (((F_CPU/1000)*w_zeropulse          )/100000)
#define w_onecycles     (((F_CPU/1000)*w_onepulse    +500000)/100000)
#define w_totalcycles   (((F_CPU/1000)*w_totalperiod +500000)/100000)
...
#define w_lowtime ((w1_nops+w_fixedlow)*100000)/(F_CPU/1000)
```
USB analyser after the modification: 
0 CODE:
time high:  0.17 µs
time low: 0.7 µs
1 CODE:
time high:  0.62 µs
time low: 0.54 µs

But why ?

cpldcpu, 2016-11-24:
This seems odd. Have you checked the listing file? (disassembly?)

neoxnitro, 2016-11-28:
Hi, thanks for your reply !

The following code was compiled in Released profiled (Optimization -Os)

`000000e8 <ws2812_sendarray_mask>:
      e8:	24 b1       	in	r18, 0x04	; 4
      ea:	24 2b       	or	r18, r20
      ec:	24 b9       	out	0x04, r18	; 4
      ee:	35 b1       	in	r19, 0x05	; 5
      f0:	24 2f       	mov	r18, r20
      f2:	20 95       	com	r18
      f4:	23 23       	and	r18, r19
      f6:	35 b1       	in	r19, 0x05	; 5
      f8:	34 2b       	or	r19, r20
      fa:	4f b7       	in	r20, 0x3f	; 63
      fc:	f8 94       	cli
      fe:	fc 01       	movw	r30, r24
     100:	68 0f       	add	r22, r24
     102:	79 1f       	adc	r23, r25
     104:	e6 17       	cp	r30, r22
     106:	f7 07       	cpc	r31, r23
     108:	81 f0       	breq	.+32     	; 0x12a <loop24+0x1c>
     10a:	81 91       	ld	r24, Z+
     10c:	98 e0       	ldi	r25, 0x08	; 8

0000010e <loop24>:
     10e:	35 b9       	out	0x05, r19	; 5
     110:	00 00       	nop
     112:	87 ff       	sbrs	r24, 7
     114:	25 b9       	out	0x05, r18	; 5
     116:	88 0f       	add	r24, r24
     118:	00 00       	nop
     11a:	00 c0       	rjmp	.+0      	; 0x11c <loop24+0xe>
     11c:	00 c0       	rjmp	.+0      	; 0x11e <loop24+0x10>
     11e:	00 c0       	rjmp	.+0      	; 0x120 <loop24+0x12>
     120:	25 b9       	out	0x05, r18	; 5
     122:	00 00       	nop
     124:	9a 95       	dec	r25
     126:	99 f7       	brne	.-26     	; 0x10e <loop24>
     128:	ed cf       	rjmp	.-38     	; 0x104 <ws2812_sendarray_mask+0x1c>
     12a:	4f bf       	out	0x3f, r20	; 63
     12c:	08 95       	ret`

closed 2016-11-28 by neoxnitro