cpldcpu/light_ws2812 #41: Examples not compiling

Kaes3kuch3n, 2017-01-10:
I just wanted to try out the library in combination with an Arduino Uno and some WS 2812 LEDs, but it won't compile correctly.
Here's the error log:

`C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp: In member function 'cRGB WS2812::get_crgb_at(uint16_t)':

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:38:35: error: 'OFFSET_R' was not declared in this scope

   px_value.r = pixels[OFFSET_R(tmp)];

                                   ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:39:35: error: 'OFFSET_G' was not declared in this scope

   px_value.g = pixels[OFFSET_G(tmp)];

                                   ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:40:35: error: 'OFFSET_B' was not declared in this scope

   px_value.b = pixels[OFFSET_B(tmp)];

                                   ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp: In member function 'uint8_t WS2812::set_crgb_at(uint16_t, cRGB)':

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:53:22: error: 'OFFSET_R' was not declared in this scope

   pixels[OFFSET_R(tmp)] = px_value.r;

                      ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:54:22: error: 'OFFSET_G' was not declared in this scope

   pixels[OFFSET_G(tmp)] = px_value.g;

                      ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:55:22: error: 'OFFSET_B' was not declared in this scope

   pixels[OFFSET_B(tmp)] = px_value.b;  

                      ^

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp: At global scope:

C:\Program Files (x86)\Arduino\libraries\light_WS2812\WS2812.cpp:61:81: error: no 'uint8_t WS2812::set_subpixel_at(uint16_t, uint8_t, uint8_t)' member function declared in class 'WS2812'

 uint8_t WS2812::set_subpixel_at(uint16_t index, uint8_t offset, uint8_t px_value) {

                                                                                 ^

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.`

The last says: "Error while compiling for the board Arduino/Genuine Uno"

Henri-K, 2017-01-14:
@Kaes3kuch3n the zipfile in the arduino folder is outdated. delete it and zip the folder within the directory instead. import this new zipfile into arduino as described in the readme.

cpldcpu, 2017-02-28:
Good point. Need to fix it some time.

manischewitzbacon, 2017-05-18:
i'm having trouble compiling examples as well - using ATTiny85 and i zipped the folder rather than using the zip in the zip.  

https://pastebin.com/MNCtfJMu

cpldcpu, 2023-11-18:
this should be fixed now

closed 2023-11-18 by cpldcpu