cpldcpu/light_ws2812 #42: My 8x8 matrix is only blinking....

LJLP16, 2017-01-13:
When I upload the Blinky example to my Arduino Uno only the first led does what it shuld do and the others are blinking in ohter colors.
Sometimes they are also just black.

Here is the code:
`/*
 * light_ws2812 example
 *
 * Created: 07.03.2014 12:49:30
 *  Author: Matthias Riegler
 */ 

#include <WS2812.h>

WS2812 LED(64); // 1 LED
	
cRGB value;

void setup() {
	LED.setOutput(7); // Digital Pin 7
}

void loop() {
	value.b = 255; value.g = 0; value.r = 0; // RGB Value -> Blue
	LED.set_crgb_at(0, value); // Set value at LED found at index 0
	LED.sync(); // Sends the value to the LED
	delay(500); // Wait 500 ms

  value.b = 0; value.g = 0; value.r = 255; // RGB Value -> Blue
  LED.set_crgb_at(0, value); // Set value at LED found at index 0
  LED.sync(); // Sends the value to the LED
  delay(500); // Wait 500 ms
	
}
`

cpldcpu, 2017-01-14:
Well, the reason for this is, that this example is only controlling one LED. if you connect more than one, it obviously does nothing. Try some of the other examples.

closed 2017-01-14 by cpldcpu

LJLP16, 2017-01-14:
ok thx.
I thought this could be the reason because the fad example works just fine... 👍