cpldcpu/light_ws2812 #44: License question: GPLv2, or GPLv2+?

algernon, 2017-02-16:
Hi!

Just a quick question about the license: is it GPLv2 only, or GPLv2 or later? By default, the GPL's text usually say " either version 2 of the License, or (at your option) any later version", and this is the recommended text to paste into comments, etc. However, the sources only say "License: GPLv2", which could be interpreted both ways.

Reason I'm asking, is because I'd like to use the library in a project that is otherwise GPL3+. If the license is GPL2+, I can, otherwise I'll have to find an alternative, which would be a huge setback.

Thanks in advance!

cpldcpu, 2017-02-16:
Good point. I never bothered much. I can change it to GPL3.

algernon, 2017-02-16:
GPL2+ is fine too, as long as the "or any later version" part applies, it's all good. No need to change to GPL3, just a clarification that this part applies is all I need =)

closed 2017-02-28 by cpldcpu