cpldcpu/light_ws2812 #45: Difference with STM32F3xxx?

rascafr, 2017-03-01:
Thanks the huge work. Works well with STM32F0 devices, but not with STM32F3xx, the assembly code is stuck into an infinite loop. 

Tried to debug the code with OpenOCD, seems like the chip is blocked at the `ilop`end instruction.

Have you an idea about where the problem may come from? Is there such a huge difference between M0 and M3 assembly code instruction and how it works?

Regards

cpldcpu, 2017-03-01:
Curious! No idea what the issue is. I have never tried it on CM3, but I believe others have. The CM3 is a superset of CM0.

One potential issue, however, is that the timing become more unpredictable with higher clocked CM3 MCUs, since there will be flash waitstates.

rascafr, 2017-03-01:
Wow, thanks a lot for the quick answer! :)
I've changed the clockspeed of the STM32F303 (72MHz to 16MHz then 8MHz) but the problem is still here.

The assembly code seems to block in the bit 1 operation (`08000d2c` → `08000d34` → `08000d38` → `08000d40` → `08000d2c` → etc.), here is a copy of the disassembly code during debug:

[image]

What do you mean by a register allocation issue? Register overridden?

Again, thanks a lot!

cpldcpu, 2023-11-18:
It's most likely an issue with flash timing and code cache. The bitbanging approach is not very suitable for cores beyond CM0.

closed 2023-11-18 by cpldcpu