cpldcpu/light_ws2812 #46: Sending RGB data to RGBW LEDs

eltang, 2017-04-15:
Is there any easy way to do this?

rascafr, 2017-04-15:
Just send a 4 byte array (eg: 0x80,0x80,0x00,0x00 for a yellow colour) using the function defined in the *. h file (ws2812_send(...)) :)

eltang, 2017-05-16:
@cpldcpu Do you have any insights?

cpldcpu, 2017-05-17:
I don't understand where the problem is. Just set W to 0?

eltang, 2017-05-17:
I'm aiming to use another library to create the data to be sent to the LEDs. The problem is that it only works which RGB data, and converting it into RGBW data every time I want to send an update takes too long. Therefore, I'm wondering if this library can insert null bytes into RGB data as it is sending it out.

cpldcpu, 2017-05-17:
The easiest way would be to copy the data to a new array - assuming you have enough memory. Otherwise you would have to build your own inner loop. That would be more involved.

closed 2017-05-25 by cpldcpu