cpldcpu/light_ws2812 #50: Red and Green Inverted

ChrisoftheBoyerClan, 2017-07-15:
Hello,

Currently, when using the AVR library with my atmega328p, my green values are inverted. For example, when using the provided blinky example, instead of going red -> green -> blue it goes green -> red -> blue.

I don't think it's a problem with the LEDs themselves because when using the FastLED library the colors come out as intended.

With some helper functions I can manually invert the red and green values again to get the expected colors, but I'd rather not have to do that.

Thoughts?

cpldcpu, 2017-07-16:
There are LEDs out there with different color orders, so it seems pretty likely that these are different.

You can easily reorder the colours by changing the struct in light_ws2812.h

`struct cRGB { uint8_t g; uint8_t r; uint8_t b; };`

ChrisoftheBoyerClan, 2017-07-18:
Awesome! Thanks!

closed 2017-07-27 by cpldcpu