cpldcpu/light_ws2812 #62: calculations based on F_CPU do not work

JBecker32, 2018-07-20:
If I set F_CPU to 8000000UL (8MHz), w1 is calculated as 0 (2-2), w2 as 3 (7-4) and w3 is **below zero** (10-8-0-3)!
This leads to a veeeeery long low period.
And, even if I set F_CPU to 500000UL (500kHz), I do not get a warning. Code is compiled without error.

cpldcpu, 2018-07-20:
Both cases are caught in the code. Have you verified this in the listing
file?

JBecker32, 2018-07-20:
But I have seen that it does not work. Period was not ok until I manually
set w3 to 0.

cpldcpu, 2018-07-21:
Can you post the listing file?

JBecker32, 2018-07-25:
Problem solved: F_CPU has to be defined as (e.g) 8000000, **not** as 8000000uL. The uL (which is needed for due to code conventions) forces the calculations to be done in integer which leads to false results.

closed 2018-07-25 by JBecker32