cpldcpu/light_ws2812 #67: Changing variables in ws2812_config does not affect pinouts

rkeithryan, 2019-02-04:
When I change ws2812_port and ws2812_pin in ws2812_config.h it has no effect on my program. The neopixel still only works on port B pin 2. I'm trying to figure out how to drive two separate strings of pixels from two different pins.

I'm using the ATMega328P, but do not want to use Arduino or the Arduino bootloader. 

My includes:
#include <util/delay.h>
#include <avr/io.h>
#include <avr/interrupt.h>
#include "ws2812_config.h"
#include "light_ws2812.h"
#define F_CPU 8000000

am I missing something here?

PaulKlinger, 2019-02-08:
I think this is because "ws2812_config.h" is not included in "light_ws2812.h" This was changed semi-recently, in 6f56c5c091ab6df631460eaa4186d4a6ded51723.

I don't really know much about C so I'm not sure if there's some way to make it use the definitions from the other include.

closed 2019-02-08 by rkeithryan

PaulKlinger, 2019-02-08:
Did you figure out a way to make it work without adding the include to "light_ws2812.h"? If not this should probably be reopened and fixed in the library.

rkeithryan, 2019-02-08:
I believe the issue was that I was not using the makefile that came with the library and the makefile I was using did not have these includes. Although I did not test this. I have since changed my direction with this project and am now using an ATTiny85. I am now trying the examples with the included makefile but have came across errors while building the library and programs. (See open issues)