cpldcpu/light_ws2812 #69: AVR configuration files do not work

PaulKlinger, 2019-02-08:
According to the readme, to set configuration options one should modify e.g. ws2812_config.h and include it before light_ws2812.h. However this does not actually work, the custom options are ignored and the default ones in light_ws2812.h are used. (See also #67.)

As far as I understand it this is because the two files are part of different compilation units and DEFINEs  are not shared between them. Before the commit 6f56c5c091ab6df631460eaa4186d4a6ded51723 ws2812_config.h was included in light_ws2812.h, so everything worked.

Is there some way to get the current setup to work or is this a bug?

EvansMike, 2019-11-14:
I can confirm that.  I added
`#include "ws2812_config.h"`
to` light_ws2812.h` to fix.

cpldcpu, 2019-12-03:
which compiler did you use?

EvansMike, 2019-12-03:
Gcc on linux, Fedora30.  Also my own Makefile.

chschu, 2020-09-08:
In 6f56c5c091ab6df631460eaa4186d4a6ded51723, the line `#include "ws2812_config.h"` has been removed from `light_ws2812.h`. Now when `light_ws2812.c` is compiled, the user-defined configuration is not included anymore.

chschu, 2020-09-08:
As a workaround, you can change the defaults in `light_ws2812.h` or pass `-Dws2812_port=...` and `-Dws2812_pin=...` as compiler flags.

cpldcpu, 2023-11-18:
This should be fixed now

closed 2023-11-18 by cpldcpu