cpldcpu/light_ws2812 #70: Attiny88 compatibility

Rainbrony, 2019-02-25:
MH-ET LIVE tiny88(16MHz) works at first burn, strap doesn't ignite after a few resets, pin 7 were used. Tested Voltage between pin 7 and GND is 3.48V. 
I'm a newbie, don't know what happened exactly

codesoap, 2019-05-11:
I've not yet used this library, but take a look at #66. The attiny85 only has 512 bytes of RAM and the Arduino bootloader probably uses up a good share of it.

codesoap, 2019-05-24:
The attiny only has 512 bytes of ram. The array for 24LEDs will take up 72 bytes already. You didn't write what else you are doing in the program, but I'd guess the RAM is used up.

Maybe you could try using raw C, so that you can easier manage your RAM usage.

What is the error you get when using more than 24 LEDs?

Rainbrony, 2019-05-24:
thanks mate

closed 2019-05-24 by Rainbrony