cpldcpu/light_ws2812 #75: Trying Xmega ATXMEGA128A1

rselec, 2019-10-13:
Hi,
I really need to have a WS2812 lib for XMEGA, so I tried compiling for XMEGA, several issues:

First the defines for DDR and PORT will not work for Xmegas
https://github.com/cpldcpu/light_ws2812/blob/68c165724458cea1ee0f257ab6627013b163d684/light_ws2812_AVR/Light_WS2812/light_ws2812.h#L94
https://github.com/cpldcpu/light_ws2812/blob/68c165724458cea1ee0f257ab6627013b163d684/light_ws2812_AVR/Light_WS2812/light_ws2812.h#L95

You need to specify something like:
`#define ws2812_PORTREG  PORTB_OUT`
`#define ws2812_DDRREG   PORTB_DIR`

But after that it won't even compile:

> impossible constraint in 'asm'	...\light_ws2812.c	119
> asm operand 2 probably doesn't match constraints	...\light_ws2812.c	119

I think it has something todo with:
https://github.com/cpldcpu/light_ws2812/blob/68c165724458cea1ee0f257ab6627013b163d684/light_ws2812_AVR/Light_WS2812/light_ws2812.c#L176
_SFR_IO_ADDR(ws2812_PORTREG) will not result in an 6-bit positive constant, it is 0x0624 (=1572) which can't fit. I tried several other constraints but I don't have any clue which one could with the register address of the XMEGA PORTB_OUT register and wether the assembler code is compatible.

Please help!

cpldcpu, 2019-12-03:
Yes, that is a general issue with XMEGA. I probably should add it to the faq.

Two possible solutions:

1) Map the port you use to the lower I/O region.
2) Use the arduino version, which is using ST to write to ports and is not limited to the lowest 64 bytes.

Jakub-Plan-d-k, 2022-06-19:
Hey, I know this is a late answer, but maybe you still will find it useful. I am using AVR128DB48 and had exactly the same issue. However, I have found that there are also VPORTB_OUT and VPORTB_DIR addresses defined. 

They work for simple LED blink and the ws2812 code compiles when you use them. Still did not get the ws2812 LED to work, but I am pretty sure this is a good direction :)

cpldcpu, 2023-11-18:
Version 2.5 should fix this issue.

closed 2023-11-18 by cpldcpu