cpldcpu/light_ws2812 #78: expected ")" error in keil

mchamrami, 2019-12-14:
hi dear
when i compile in keil ide ...show this error

error:  #18: expected a ")"

cpldcpu, 2019-12-14:
Which file exactly?

Well, actually the inline assembly uses AVR-GCC format. It is quite possible that there is an issue with Keil. I personally never tried Keil, but maybe someone else had.

closed 2020-02-17 by cpldcpu

QDMengChong, 2025-03-20:
@cpldcpu I think you are right. I also have the same problem when I use file `/light_ws2812/light_ws2812_ARM/light_ws2812_cortex.c` on https://github.com/cpldcpu/light_ws2812/blob/f214bf01c9ef2426097e5c42274b84451a7240e4/light_ws2812_ARM/light_ws2812_cortex.c#L111 on keil. I never used inline assembly, so it's very difficult for me to fix it. Can you give me some help?