cpldcpu/light_ws2812 #82: Need help

bokolob, 2020-10-04:
I use your library with AtTiny85 (8Mhz internal oscillator). 
Simple loop 

```
    for (int i = 0; i < 100; i++) {
        led[0].r=GREEN[i][0];
        led[0].g=GREEN[i][1];
        led[0].b=GREEN[i][2];
        led[0].w=0;

        ws2812_setleds_rgbw(led,1);
        _delay_ms(50);
    }
```

GREEN - is a predefined array with RGB values. Only green is non-zero here. 
So, I've got very strange artifacts here, sometimes led blinks red, and white. It has to start from black and slowly go green. But it starts at green instead.  It's 100% reproducible and probably goes wrong at the same GREEN rows.

At the same time, it works perfectly on the Arduino Uno board.

I think that something is wrong with timings. But I have no idea how to investigate 8(

bokolob, 2020-10-04:
Oh, yes. I use a single led.

cpldcpu, 2020-10-04:
At 8MHz the timing of the library is somewhat marginal. Have you tried a higher clock frequency?

bokolob, 2020-10-05:
All pins are used, so I have no space for Crystal ;) As far as I understand 8 MHz is highest speed for an internal oscillator.

But I can’t say that it doesn’t work at all, It just has some glitches. I think it depends on amount of sending 1 bits.

bokolob, 2020-10-05:
So I tried arduino uno, which has 16 MHz clock. And it worked well.

cpldcpu, 2020-10-05:
You can set ATtiny85 to 16MHz with internal oscillator using the PLL.

closed 2021-03-29 by cpldcpu