cpldcpu/light_ws2812 #84: Impossible constraint in 'asm'

azkara22, 2020-12-23:
I am trying to use the ws2812 code for AVR with a ATtiny3217 but I get "Impossible constraint in 'asm'" error along with "asm operand 2 probably doesn't match constraints" warnings. The error looks like it's coming from the following instruction, but I can't get it work.

`"I" (_SFR_MEM_ADDR(ws2812_PORTREG)`

cpldcpu, 2020-12-23:
Did you define a port register that is part of the I/O region?

azkara22, 2020-12-23:
I have just found that the PORTC register that I am using is out of the I/O region. The solution in my case has been to use the VPORTC_OUT register that it is on that region. Thank you for the response anyway.

Sponge5, 2021-04-03:
Hi, is there a way to use a register outside the I/O region? In my case I need to use PH5 on ATMega2560. In `iomxx0_1.h` PORTH is defined as `_SFR_MEM8(0x102)`.

Sponge5, 2021-04-03:
The way I understand this is to replace `out` instructions with `sts` and fiddle with the amount of `noop`'s. I'll probably resort to rewiring the board 😅

cpldcpu, 2021-04-04:
Have a look at the Arduino version. It actually uses sts for port access.

https://github.com/cpldcpu/light_ws2812/blob/master/light_ws2812_Arduino/light_WS2812/light_ws2812.cpp

cpldcpu, 2023-11-18:
Version 2.5 should fix this issue.

closed 2023-11-18 by cpldcpu