cpldcpu/light_ws2812 #89: Licensing of your code

rainerschoe, 2021-03-27:
Hi, thanks for implementing this :)
I would like to use your library in an AVR Project of mine.
I read your license (GPLv3), however I do not understand how this is intended to apply to my project:
If I compile your lib (without modifications) into object files and link them into my project, do I need to publish my whole source code?

I am more used to open-source licenses like Apache2, BSD or LGPL, which explicitly allow linking. It it possible to re-license your lib in one of those?

cpldcpu, 2021-03-29:
Hi, do you want to use this code for a commercial project? If that is not the case, then the license should not be an issue. If you don't publish your project in any way, you also do not need to publish the sources according to the GPL. It's very simple.

rainerschoe, 2021-03-29:
Thanks for your reply.

Right now it is a private hobby project, but I wanted to keep my options open to either publish as open-source or as a commercial product in the (far) future.

I will use your code for now as is then, and whenever I want to publish in a different license than GPL I will contact you again or write my own implementation :)

closed 2021-03-29 by rainerschoe