cpldcpu/light_ws2812 #92: WS2812B with BeagleBone Black

rajmehta28599, 2021-07-06:
Aim: Control WS2812B Led Strip with BeagleBone Black in C language.

Thank you for your contribution. I read your readme file. you also provide c code but related to AVR boards can you help me with changes related to BeagleBone Black(Linux). 

Thank you.

cpldcpu, 2021-07-28:
I don't think bitbanging, as used in this library, is the right approach on the beagle bone black. I am certain there are SPI based solutions out there.

closed 2021-07-28 by cpldcpu

ole00, 2022-08-10:
Beagleboe has 2 PRU Units - basically 2 small MCU's that can do the bitbanging independently on Linux. There is an interface that can feed the PRU's with the data (LED colours) into 8kbyes shared buffer between Linux userspace (mmapped memory area) and a PRU. Just a thought, maybe that's not the most practical solution though.