cpldcpu/light_ws2812 #98: Compile error: DDRB is not declared (AVR)

ygoe, 2022-09-01:
The code won't compile. The symbol `DDRB` is not declared. I don't know what that is.

I'm using an ATtiny1614 chip. I've used registers before but never seen that one.

cpldcpu, 2022-09-01:
Have you changed the port names in the config file? DDRB is decleard in the default file, you probably have to adapt it to your MCU.

ygoe, 2022-09-02:
It's an empty project, I haven't changed any names. I wouldn't know how to do so.

I guess that DDRB exists on older AVR MCUs but not the newer 1-series like my ATtiny1614. So the code is probably incompatible with these AVR devices without modifications.

Meanwhile I figured out that DDRB sets the pin direction (input or output). That works differently on my chip. But I can't follow the ASM code where it's used so I can't fix that.

It's not time-critical for me anymore as I managed to write my own driver for this protocol. It wasn't especially hard, just plain C code, setting the pin registers, and the `__builtin_avr_delay_cycles()` function for exact timings. At 20 MHz there's plenty of time.

simwai, 2023-11-13:
I have the same problem with my XMEGA A1 Xplained with atxmega128a1. Don't understand how to work around this.

cpldcpu, 2023-11-16:
Thats and issue with the port location, you have to remap the port. The library tries to access the I/O ports using the out instruction.

Instead, you could also try the arduino version which should not show the described issue.

ygoe, 2023-11-16:
@simwai Here's some files from one of my experiment projects. I haven't touched it in a while but it should work. Ignore the extra parts about additional LEDs and the button switch, they're on my breadboard for debugging purposes and to play with.

The interesting part is in ws2813.cpp and ws2813.h and there's some demo code in the functions in main.cpp.

[AvrTest1.zip](https://github.com/cpldcpu/light_ws2812/files/13380518/AvrTest1.zip)

In case of compiler issues, I use VSCode with Platform IO IDE. From platformio.ini:

```ini
[env:ATtiny1614]
platform = atmelmegaavr
board = ATtiny1614
;no framework
```

cpldcpu, 2023-11-18:
The library should now support ports that are outside the I/O space.

Please try with v2.5

closed 2023-11-18 by cpldcpu

simwai, 2023-11-18:
Thanks a bunche, mate! 🙌

cpldcpu, 2023-11-19:
No problem! 😀 

Let me know if it works, please. I could only test on ATMega168.

simwai, 2023-11-21:
@cpldcpu It didn't work. I use the AVR version and it seems like nothing changed there (at least nothing visible for me). The problem is that XMEGA A1 XPLAINED doesn't have DDRB in the <avr/io.h> file.

cpldcpu, 2023-11-21:
Indeed, it appears the registers have been renamed. Unfortunately I don't have an XMEGA to test. But it looks like the functionality of the port registers is still unchanged, so when replacing with the correct port names, the library shoudl still work.

[image]
[image]

simwai, 2023-11-21:
I tried to replace DDRB by DIR, but this led to unwanted side effects in the bit manipulation operation.

cpldcpu, 2023-11-21:
Not sure I understand? Do you have an example / error message etc?

simwai, 2023-11-23:
@cpldcpu Unfortunately, I didn't record it, but when I get one again, I post it here. Thank you for your help anyways. :)