cpldcpu/light_ws2812 #100: Blink example works for when LED(112) or smaller but not larger.

acroscarrillo, 2022-11-16:
Blink example works on ATTINY84 (turns on first LED) if `WS2812 LED(112);` or smaller, that is `WS2812 LED(1);`, `WS2812 LED(2);`, ... ,  `WS2812 LED(112);`. However, whenever  `WS2812 LED(113);` or higher it either doesn't light any LEDs or it lights up multiple random LEDs in my 16x16 array with random colours. This is not a current draw problem as I can light up 112 LEDs if I want to. Further, the code works perfectly if uploaded to an Arduino Uno so the issue seems to be in the interplay between the ATTINY84 and the library. Here's the code:
`/*
 * light_ws2812 example
 *
 * Created: 07.03.2014 12:49:30
 *  Author: Matthias Riegler
 */ 

#include <WS2812.h>

WS2812 LED(113); // 1 LED
	
cRGB value;

void setup() {
	LED.setOutput(7); // Digital Pin 9
}

void loop() {
	value.b = 1; value.g = 0; value.r = 0; // RGB Value -> Blue
	LED.set_crgb_at(0, value); // Set value at LED found at index 0
	LED.sync(); // Sends the value to the LED
	delay(500); // Wait 500 ms
	
	value.b = 0; value.g = 0; value.r = 1; // RGB Value -> Red
	LED.set_crgb_at(0, value); // Set value at LED found at index 0
	LED.sync(); // Sends the value to the LED
	delay(500); // Wait 500 ms
}
`

acroscarrillo, 2022-11-29:
Hmm, reading further it seems like I might be running out of "SRAM". So even though my code compiles and loads correctly with no memory issues on the Arduino IDE, somehow the ATtiny84 runs out of this "SRAM" memory and so the board behaves erratically. I'm not 100% sure this is the issue but I think it is. I might be moving to another library more memory efficient where I can declare smaller memory slots per LED as I dont have to use many colors (just 2-3) and one single level of brightness.

cpldcpu, 2022-11-29:
Yes, I am pretty sure that is related to that. The ATtiny85 has significantly smaller SRAM than the ATMega328. 

The LED data is stored in the SRAM before it is sent out to the WS2812 chain. When there are too many LEDs, there is no memory left and issues occur. Each LED requires 3 bytes of memory.

Solution: Use MCU with more SRAM. Alternatively you could also use a loop to send out the same content several times, but then your pattern generation is limited.

closed 2022-11-29 by cpldcpu