cpldcpu/light_ws2812 #101: inline assembly not recognized

orotarobas, 2022-12-22:
We're trying to build the project for LPC812 using IAR compiler but we get errors since the assembly instructions are not recognized.
What toolchain are we supposed to use in order to build the project for the LPC81x family?

cpldcpu, 2022-12-22:
I used the NXP supplied IDE, which is based on GCC as far as I know. It's quite possible that the format of the inline assembler is different. Maybe it is possible to adjust the source?

orotarobas, 2022-12-23:
is it the MCUXpresso (Netbeans based)?

Would you be so kindo to send the project or put it on github?

Bests

Lorenzo

------ Messaggio originale ------
Da "Tim" ***@***.***>
A "cpldcpu/light_ws2812" ***@***.***>
Cc "orotarobas" ***@***.***>; "Author" 
***@***.***>
Data 22/12/2022 19:14:46
Oggetto Re: [cpldcpu/light_ws2812] inline assembly not recognized (Issue 
#101)

closed 2023-11-19 by cpldcpu