cpldcpu/light_ws2812 #102: undefined reference to `__heap_end'

NickkoG, 2023-02-06:
Hello, I use Arduino 2.0.2 (lastest build) and I have imported the light_WS2812 ZIP file.
I try to complie the example but I get this error:

`/home/nickko/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -flto -fno-reorder-blocks -ffixed-r2 -ffixed-r3 -ffixed-r4 -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10607 -DARDUINO_attiny -DARDUINO_ARCH_AVR -I/home/nickko/.arduino15/packages/MicroCore/hardware/avr/2.2.1/cores/microcore -I/home/nickko/.arduino15/packages/MicroCore/hardware/avr/2.2.1/variants/attiny13 -I/home/nickko/Arduino/libraries/light_WS2812 /home/nickko/Arduino/libraries/light_WS2812/light_ws2812.cpp -o /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/libraries/light_WS2812/light_ws2812.cpp.o
Compiling core...
Using precompiled core: /tmp/arduino-core-cache/core_MicroCore_avr_13_clock_9M6,eeprom_keep,BOD_2v7_4d3f75b13a46ca2a177c83ea51e02333.a
Linking everything together...
/home/nickko/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -w -Os -flto -fno-reorder-blocks -ffixed-r2 -ffixed-r3 -ffixed-r4 -g -fuse-linker-plugin -Wl,--gc-sections -mrelax -mmcu=attiny13a -nostartfiles -Wl,--undefined=__vectors -o /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/sketch_feb6b.ino.elf /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/sketch/sketch_feb6b.ino.cpp.o /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/libraries/light_WS2812/WS2812.cpp.o /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/libraries/light_WS2812/light_ws2812.cpp.o /tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135/../arduino-core-cache/core_MicroCore_avr_13_clock_9M6,eeprom_keep,BOD_2v7_4d3f75b13a46ca2a177c83ea51e02333.a -L/tmp/arduino-sketch-0E23F795420FFB8FF1B27C4DE58C4135 -lm
/home/nickko/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr25/tiny-stack/libc.a(malloc.o):(.data+0x0): undefined reference to `__heap_end'
collect2: error: ld returned 1 exit status`

cpldcpu, 2023-11-18:
I updated the zip file - i hope its fixed now?

ihatechoosingusernames, 2024-01-28:
Not fixed for me, compiling for ATTiny13A on Arduino IDE 2.2.1

I imported the zip file and tried to compile the blinky example with updated pin.

```
.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr25/tiny-stack/libc.a(malloc.o):(.data+0x0): undefined reference to `__heap_end'
collect2: error: ld returned 1 exit status
```

cpldcpu, 2024-05-01:
I believe it is running out of memory on the ATiny13A. Just exclude that example in the Makefile.