cpldcpu/light_ws2812 #103: Request to add a tag for v2.4

U-1F992, 2024-04-30:
Thank you for developing this useful library. I have been relying on it, particularly for projects involving the ATtiny10. While I was little disappointed to see support for the ATtiny10 dropped in v2.5, I understand the challenges on maintaining support for such constrained hardware as well as the necessity to enhance performance for other newer devices.

I noticed that [v2.4](https://github.com/cpldcpu/light_ws2812/tree/cd149996012fe96bc3d7883cb18a0103fd8e8b3a) does not have a corresponding git [tag](https://github.com/cpldcpu/light_ws2812/tags). This version will be crucial for those of us using some devices, and currently, we have to refer to the commit directly as per the README. Adding a tag for v2.4 would greatly simplify access to this important version and help ensure stable builds for our projects.

Could you please consider adding a git tag for v2.4? This would be immensely helpful for those of us needing to maintain compatibility.

cpldcpu, 2024-05-01:
Good point, the update also broke compatibility for the reduced AVR core. 

And indeed, I forgot to tag v2.4. But at this point it appears difficult to do so, becuase a lot of smaller additions have been submitted, so it is difficult to pinpoint.

I think it may be easier, just to add compatibility back in for that ATtiny10. Would you be willing to test?

U-1F992, 2024-05-01:
Of course. It may take a little time to follow up, but I will try.

cpldcpu, 2024-05-01:
There you go:
https://github.com/cpldcpu/light_ws2812/tree/v2.6testing

Please let me know if it works on-device, I will release it then.

U-1F992, 2024-05-06:
@cpldcpu Sorry it took so long. It works properly with ATtiny10. https://github.com/U-1F992/attiny10-neopixel

cpldcpu, 2024-05-06:
No problem, there is no rush. Thank you very much! Will merge.

closed 2024-05-06 by cpldcpu