cpldcpu/light_ws2812 #110: tiny85 wont meet timing

marek-honek, 2025-04-05:
Hi, I'm using an ATtiny85 with fuses set to internal 8 MHz clock (E:FF, H:DF, L:E2) and F_CPU defined as 8000000UL. WS2812 LEDs only light up white. Your library and examples output pulses ~4.2 µs wide instead of sub-µs (read on scope). Any idea why? Thanks!

cpldcpu, 2025-04-07:
Usually that is an issue with the MCU clock being lower than expected and unrelated to the library,

Did you try testing with some kind of loop that toggles a pin? Or did you try to provided demo?

marek-honek, 2025-04-08:
I did try to set the L fuse to 0xA2 to enable CLKO and saw 8 MHz square wave, so I assume the system clock frequency is not the issue. I tried many things, toggling the pin in assembler also indicates the chip is running on 8 MHz. Once I am at home, I'll post additional info.